IHP-GmbH/IHP-Open-PDK #523: Differential Simulation using OpenEMS

rupokdas147, 2025-05-28:
I have tried to design a Differential Coupled Line coupler at 155 GHz. I have calculated the line width and spacing using the equations from the publication referenced in [this link](https://ieeexplore.ieee.org/abstract/document/1643581).
I found that my calculated width and spacing are the same as described in the publication referenced in [this link](https://ieeexplore.ieee.org/abstract/document/9737284)
Therefore, I designed the coupler with the exact width, spacing, and metal layers specified in the second publication, assuming they have a similar metal stack in their PDK. The authors of the second publication used ADS to simulate the coupler.
In OpenEMS, we do not have a differential port like ADS. So, I have used 8 single-ended ports instead of 4 differential ports (used in ADS). Then, I have converted the 64 s-parameters into 64 mixed-mode s-parameters using the sk-rf Python library and observed the differential S21 and differential S31 curves. The curve is not similar to the actual coupler, not even nearly similar. I have tried various line widths and distances. I have also tried ring ring-type GND layer as well as a solid metal GND layer. No method shows the output as an actual coupler.
Then I tried TopMetals to design the coupler and kept both of them very close (about 1 μm apart). The curve was slightly close to a coupler output shape.
My question is: The method I am using to observe the differential s-parameters, is it the right way to determine them? Is it possible to design a differential coupler in OpenEMS?
I am attaching my test GDS file and Python code. 

[Coupled_Line_Coupler_Diff.zip](https://github.com/user-attachments/files/20481284/Coupled_Line_Coupler_Diff.zip)

[image]

rupokdas147, 2025-05-29:
@VolkerMuehlhaus Thanks for reviewing the issue. You may look at this publication as it's open access: https://ieeexplore.ieee.org/document/10061405
There is a quadrature coupler design in this publication. They are also using the same metal stack or the same process I guess. 
I have looked into your transformer differential simulation and tried this method also. Though I am getting the 90 degree phase shifted signal properly in both method mixed mode and the method used in your transformer method. But not getting the coupled signal properly. The magnitude of the coupled signal is very low (less than -40 dB) where I designed for -3 dB coupling. 

So, I am assuming that the signal is not coupling properly through the dielectric between the conductors.

rupokdas147, 2025-05-29:
@VolkerMuehlhaus Thanks. Can you please elaborate more about the problem happening with my simulation. I am trying to recognize the issue for a long time.

VolkerMuehlhaus, 2025-05-29:
@rupokdas147 Just look carefully at Fig. 14 of the reference paper, to see what each differential line is.

You will notice that their differential lines are "crossover". Let me try once more to show you what they do differently:

[image]

closed 2025-05-30 by rupokdas147