IHP-GmbH/IHP-Open-PDK #527: Magic does not extract Base of pnpMPA

devstdin, 2025-05-30:
Extracting the attached layout of a `pnpMPA` transistor in magic with the following commands:
```
ext2spice lvs
extract all
ext2spice
```
... produces the below subcircuit with the Base-connection missing:
```
.subckt t_test_flat E B C
	X0 C w_338_n923# E pnpMPA a=4p p=8u
.ends
```
Adding `pnp` to the connection-section of the magic technology-file as follows:
```
connect
  *nwell,*nsd,*hvnsd,dnwell,pnp *nwell,*nsd,*hvnsd,dnwell,pnp
```
... leads to a correct extraction:
```
.subckt t_test_flat E B C
	X0 C B E pnpMPA a=4p p=8u
.ends
```
Discussion in [FOSSi-chat](https://matrix.to/#/!eXntrxjkBFnWmSjXdA:fossi-chat.org/$cb5LRiS68QHVkTbzG3mW5Y8pOADarQxdA0KJJg4bphY?via=fossi-chat.org&via=matrix.org&via=underzsea.com)

[t_test_flat.mag.txt](https://github.com/user-attachments/files/20522226/t_test_flat.mag.txt)

sergeiandreyev, 2025-10-01:
Hi @RTimothyEdwards, could you please take a look?

RTimothyEdwards, 2025-10-01:
@devstdin :  Thanks for the report and suggested fix.  I will add that to the stuff I am currently working on (mainly for extraction) and should get a commit out by the end of the day (Friday worst case).

sergeiandreyev, 2025-11-03:
Hi @RTimothyEdwards, do you have an update on this issue?