IHP-GmbH/IHP-Open-PDK #542: CI: Cache KLayout

mole99, 2025-06-13:
All of the LVS tests download KLayout from https://www.klayout.org/downloads/Ubuntu-24/klayout_0.29.8-1_amd64.deb individually.

This sometimes fail for some of them, causing the CI to get stuck:

```
--2025-06-13 07:10:43--  https://www.klayout.org/downloads/Ubuntu-24/klayout_0.29.8-1_amd64.deb
Resolving www.klayout.org (www.klayout.org)... 178.77.102.107
Connecting to www.klayout.org (www.klayout.org)|178.77.102.107|:443... failed: Connection timed out.
Retrying.
--2025-06-13 07:12:58--  (try: 2)  https://www.klayout.org/downloads/Ubuntu-24/klayout_0.29.8-1_amd64.deb
Connecting to www.klayout.org (www.klayout.org)|178.77.102.107|:443... failed: Connection timed out.
Retrying.
--2025-06-13 07:15:14--  (try: 3)  https://www.klayout.org/downloads/Ubuntu-24/klayout_0.29.8-1_amd64.deb
Connecting to www.klayout.org (www.klayout.org)|178.77.102.107|:443... failed: Connection timed out.
Retrying.
...
```

This ultimately leads to a timeout and failure of the test.

Instead of downloading KLayout for each test, it should be cached for the next run.

See here for more: https://docs.github.com/en/actions/writing-workflows/choosing-what-your-workflow-does/caching-dependencies-to-speed-up-workflows

sergeiandreyev, 2025-06-13:
Hi @FaragElsayed2, if I remember correctly, you mentioned you optimized this in one of the DRC related PRs?

FaragElsayed2, 2025-06-15:
Hi @sergeiandreyev,

Yes, I've combined the LVS device checks (unit LVS tests) into a single job. The workflow now runs four jobs in total:

- **Linting**  
- **Unit DRC Regression**  
- **Unit LVS Regression**  
- **Standard Cells LVS Regression**

To reduce the connection issues, I added some retry and timeout options to the KLayout download step.

However, I agree with this suggestion, it is more reliable solution. I supported it now in #529

Thanks @mole99 for the suggestion!

CC @atorkmabrains

sergeiandreyev, 2025-07-15:
Hi @mole99, the update to the LVS regression flow is now merged as https://github.com/IHP-GmbH/IHP-Open-PDK/pull/567, did you have a chance to check?

mole99, 2025-07-16:
Hi @sergeiandreyev, yes #567 seems to add the caching logic to the LVS regression, thanks!
However, I think #529 is still needed to cache KLayout for the DRC regressions.

FaragElsayed2, 2025-07-17:
Hi @mole99 
The DRC regression already supports this, just like the LVS regression.

closed 2025-09-01 by sergeiandreyev