IHP-GmbH/IHP-Open-PDK #543: xschem: add upper limits for L in mosfets dimensions check

mole99, 2025-06-16:
The xschem mosfets dimension checks only check wether the length of the transistors is above the minimum length: https://github.com/IHP-GmbH/IHP-Open-PDK/blob/68eebafcd9b2f5e92c69d37a8d3d90eb266550f5/ihp-sg13g2/libs.tech/xschem/xschemrc#L457

However, the valid range for the model also lists 10um as the maximum value for the length: https://github.com/IHP-GmbH/IHP-Open-PDK/blob/082805f51d3e59f23ef458c4120083c23f52f393/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_moslv_parm.lib#L56

This applies to both the lv and hv mosfets.
It would be great if the xschem dimension checks also included the upper limits.

KrzysztofHerman, 2025-06-20:
@mole99 please checkout this functionality PR #554. It implements the checks.

mole99, 2025-06-22:
I tested your branch and it works as expected: Both lv and hv mosfets now display a warning if L > 10.
Thanks!

sergeiandreyev, 2025-06-27:
@KrzysztofHerman, but this one #554 was not merged?