IHP-GmbH/IHP-Open-PDK #550: FFT Output Interpretation Issues: Resolution Effects, Frequency Shifts and Magnitude Scaling,

Atif07Ali, 2025-06-20:
System Information:
qucs-s version: 25.1.2
ngspice version: 44

**1. FFT Magnitude Varies with Frequency Resolution (df)**
I've observed that changing the FFT frequency resolution (df) from 10 MHz to 1 MHz causes a significant variation in the output magnitude, even though the input signal remains unchanged.
Could you explain how the FFT magnitude depends on df?

[image]

[image]

**2. FFT Peak Shifts When Bandwidth (BW) is Narrowed**
When I reduce the FFT analysis bandwidth (BW) to below the input signal frequency (140 GHz), the FFT peak unexpectedly appears at other frequencies within the selected range, instead of remaining at 140 GHz. Since the input is a pure 140 GHz sinusoid, there should not be any frequency spike except at 140 GHz. The appearance of peaks at other frequencies is unexpected.
Could you give some insights into this behavior?

[image]

[image]

**3. Clarification Needed on FFT Magnitude Scaling**
When applying a 1 mV peak sinusoidal input at 140 GHz, the resulting FFT magnitude is approximately 902 μV. This value does not correspond clearly to either the peak or RMS of the input signal. Could you please clarify how the FFT magnitude is computed and whether it represents the peak value, RMS, or a window-scaled quantity? Understanding this would help interpret the results more accurately.

[image]

dwarning, 2025-06-21:
This is not an PDK issue.
I would propose to open a case in [ngspice users discussion forum](https://sourceforge.net/p/ngspice/discussion). But simple spice files w/o unneeded library entries are preferable.

Atif07Ali, 2025-06-21:
@dwarning I have created a ticket on SourceForge