IHP-GmbH/IHP-Open-PDK #593: Request to update the layout parameterised cells in ihp 130nm

DiarmuidSLICE, 2025-07-13:
Hi,

At present the parameterised cells (p-cells) do not include base metal layers. They go to gate poly layer only.
This means M1 strips need to be drawn over each S/D. In addition, the poly gate extensions need to be created
to enable hook up to the gates. This can be extremely time consuming when trying to keep everything DRC clean.

Request is therefore to have p-cells where you can specify the following:

- Metal you want to route into the S/D (e.g. you specify M2, then M1 + M2 strips appear across the S/D regions)
- Gate poly extensions should be included by default

This is typically what is offered in industry and is most likely a small edit to the existing p-cells (which would have a 
massive impact in terms of time saving). 

Thanks,

Diarmuid

sergeiandreyev, 2025-07-15:
Hi @DiarmuidSLICE, thank you, now it's under internal review..

DiarmuidSLICE, 2025-07-16:
Yes @sergeiandreyev . I am referring to MOSFET Pcells. Let me know if you would like me to prepare slides to illustrates things more.

DiarmuidSLICE, 2025-07-19:
Great. Im on holidays right now. Back July 28th so will email slides later that week. Probably no point doing any further work on this until I send the slides as they will specify exactly what will work best.

DiarmuidSLICE, 2025-08-01:
Just uploaded the ppt there now. It explains things in good detail but if you feel a meeting would also be beneficial, please feel free to schedule one. 

What sort of effort is this request and if accepted, what would be a realistic timeline for it? 

Thanks,

DiarmuidSLICE, 2025-08-08:
Great! Looking forward to the outcome!

DiarmuidSLICE, 2025-12-01:
Is anyone working on this?