IHP-GmbH/IHP-Open-PDK #594: Polarity in ESD-Devices is wrong in LVS

EngGhaith, 2025-07-15:
Hi everybody!

Xschem:

[image]

Netlist:

```
.SUBCKT TESTESD VDD
*.PININFO VDD:B
D1 sub! VDD nmoscl_4 m=2
.ENDS
.GLOBAL sub!

```
Klayout LVS:

[image]

Files: (Xschem+Klayout)

[TESTESD.zip](https://github.com/user-attachments/files/21229534/TESTESD.zip)

Extracted netlist:
```
.SUBCKT TESTESD VDD SUB!
D$1 SUB! VDD nmoscl_4 m=2
.ENDS TESTESD
```

Description:
Although the test for this test case is LVS pass, but it says that the pins should be flipped.
In bigger circuit would the test fail. Here it is OK because we do not have any other components tied to these nets.
Please consider that by netlist generation / LVS extraction to be repaired.

Thank you in advance.
Ghaith.

d-m-bailey, 2025-07-15:
@KrzysztofHerman @FaragElsayed2 
Actually, the problem might be in the `xschem` [symbol](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.tech/xschem/sg13g2_pr/nmoscl_2.sym) or `ngspice` model. The xschem symbol has the ports in `vss, vdd` order - standard diode order annode, cathode.
```
B 5 -2.5 27.5 2.5 32.5 {name=vss dir=inout propag=1}
B 5 -2.5 -32.5 2.5 -27.5 {name=vdd dir=inout pinnumber=2}
```

However, in the [ngspice simulation models](https://github.com/IHP-GmbH/IHP-Open-PDK/blob/main/ihp-sg13g2/libs.tech/ngspice/models/sg13g2_moslv_mod.lib), the ports are reversed.
```
* ESD MOS Clamp devices
*****************************************************************
.subckt nmoscl_2  VDD VSS
+l=0.36u w=168u m=1 ng=1 trise=0  as = 58e-12  ad = 58e-12 pd = 336u ps = 336u rfmode=0 pre_layout=1

.include sg13g2_moslv_parm.lib

 Nsg13_lv_pmos VDD VSS VSS VSS nmoscl_psp w='w' l='l' nf='ng' as='as' ad='ad' pd='pd' ps='ps' mult='m'  dta=trise ngcon=2
.ends nmoscl_2

.subckt nmoscl_4  VDD VSS
+ l=0.36u w=168u m=1 ng=1 trise=0 as = 114e-12  ad = 114e-12 pd = 672u ps = 672u rfmode=0 pre_layout=1
.include sg13g2_moslv_parm.lib

Nsg13_lv_pmos VDD VSS VSS VSS nmoscl_psp w='w' l='l' nf='ng' as='as' ad='ad' pd='pd' ps='ps' mult='m'  dta=trise ngcon=2
.ends nmoscl_4          
```

sergeiandreyev, 2025-07-16:
@d-m-bailey, thank you for looking into this!
the port order of these devices should be "VDD VSS" (it relates to simulation models, netlists, etc.)

KrzysztofHerman, 2025-08-26:
@EngGhaith @d-m-bailey  the issue was solved in #600. Please note that the diode symbol binds to `nmos` device in clamp configuration so it does not follow the (D anode cathode dmodel) pin order convention.

PR #600 (2025-08-26): nmos_cl symbols for xschem updated
Fixes #594 
Port order changed for `xschem` symbols.