IHP-GmbH/IHP-Open-PDK #626: Simulation Issue with sg13g2_sdfbbp_1: Cannot Test Falling Edge & Potential Pin Order mismatch

adarshvp123, 2025-08-04:
Hello,

I am reporting a significant issue when simulating the sg13g2_sdfbbp_1 standard cell in ngspice. The primary problem is that it seems impossible to test a CLK->Q falling edge (1 -> 0) transition . Furthermore, an incorrect pin order does produce a transition,

**Main Issue: Falling Edge Test Fails with Correct Pin Order**

When using the pin order specified in the .subckt definition (Q Q_N CLK D...), the cell appears to initialize with Q=0 and cannot be forced into a Q=1 state to begin the test. Both Q and Qn remains same in this case

Testbench with Correct Pin Order:
This testbench uses the correct positional mapping from the .subckt line. It attempts to test a 1 -> 0 transition.

```
Generated spice
* ================================================================
* TESTBENCH 1: CORRECT PIN ORDER
* ================================================================
.lib 'cornerMOSlv.lib' mos_tt
.include 'sg13g2_stdcell.spice'
.global VDD VSS

* --- Parameters ---
.param VDD_VAL=1.5, C_LOAD=64.8f, CLK_SLEW=0.3294n, D_SLEW=0.3294n
.param CLK_START_TIME=5.0e-9, delta_ns=0.5

* --- Power & DUT ---
VVDD VDD 0 DC {VDD_VAL}
VVSS VSS 0 DC 0
* Pin Order: Q Q_N CLK D RESET_B SCD SCE SET_B VDD VSS
Xdut n_q n_qn n_clk n_d n_reset_b n_scd n_sce n_set_b VDD VSS sg13g2_sdfbbp_1

* --- Stimuli for 1->0 test ---
V_CLK n_clk VSS PULSE(0 {VDD_VAL} {CLK_START_TIME} {CLK_SLEW} {CLK_SLEW} 10n 20n)
V_D n_d VSS PULSE({VDD_VAL} 0 {CLK_START_TIME-(delta_ns*1e-9)} {D_SLEW} {D_SLEW} 10n 20n)
V_RESET_B n_reset_b VSS DC {VDD_VAL}
V_SET_B n_set_b VSS DC {VDD_VAL}
V_SCE n_sce VSS DC 0
V_SCD n_scd VSS DC 0

* --- Loads & Analysis ---
CLoad_Q n_q VSS {C_LOAD}
CLoad_QN n_qn VSS {C_LOAD}
.options TEMP=25 TNOM=25 METHOD=TRAP GMIN=1e-15 RELTOL=1e-3
.tran 10p 10n uic
.end

```

Result: With this correct wiring, the simulation shows Q starts at 0 and stays at 0. No transition is observed. We believe this is because the necessary uic flag forces a Q=0 initial state, and all attempts to override this with asynchronous set pulses have failed, suggesting a startup race condition.

**Secondary Issue: Incorrect Pin Order Produces a Transition**

 if we rearrange the pins in the Xdut instantiation to a different order (which happens to be roughly alphabetical), a transition is observed.

Testbench with Incorrect (Alphabetical-like) Pin Order:
This testbench uses an incorrect pin order but is the one that produces an output transition.

Generated spice
```
* ================================================================
* TESTBENCH 2: INCORRECT (ALPHABETICAL-LIKE) PIN ORDER
* ================================================================
.lib 'cornerMOSlv.lib' mos_tt
.include 'sg13g2_stdcell.spice'
.global VDD VSS

* --- Parameters ---
.param VDD_VAL=1.5, C_LOAD=64.8f, CLK_SLEW=0.3294n, D_SLEW=0.3294n
.param CLK_START_TIME=5.0e-9, delta_ns=0.5

* --- Power & DUT ---
VVDD VDD 0 DC {VDD_VAL}
VVSS VSS 0 DC 0
* Pin Order: CLK D Q Q_N RESET_B SCD SCE SET_B VDD VSS (Incorrect)
Xdut n_clk n_d n_q n_qn n_reset_b n_scd n_sce n_set_b VDD VSS sg13g2_sdfbbp_1

* --- Stimuli for 1->0 test ---
V_CLK n_clk VSS PULSE(0 {VDD_VAL} {CLK_START_TIME} {CLK_SLEW} {CLK_SLEW} 10n 20n)
V_D n_d VSS PULSE({VDD_VAL} 0 {CLK_START_TIME-(delta_ns*1e-9)} {D_SLEW} {D_SLEW} 10n 20n)
V_RESET_B n_reset_b VSS DC {VDD_VAL}
V_SET_B n_set_b VSS DC {VDD_VAL}
V_SCE n_sce VSS DC 0
V_SCD n_scd VSS DC 0

* --- Loads & Analysis ---
CLoad_Q n_q VSS {C_LOAD}
CLoad_QN n_qn VSS {C_LOAD}
.options TEMP=25 TNOM=25 METHOD=TRAP GMIN=1e-15 RELTOL=1e-3
.tran 10p 10n uic
.end

Result: This simulation produces an output transition. 

This conflicting behavior leads to two main questions:

Pin Order Confirmation: Can you please confirm that the pin order in the .subckt definition (Q Q_N CLK D...) is indeed the correct, intended order for instantiation? The fact that an alphabetical-like order produces a result is confusing.

Recommended Initialization Procedure: Given the simulation difficulties, is there a recommended procedure or a reference testbench for reliably initializing the sg13g2_sdfbbp_1 cell to a Q=1 state? This is necessary to characterize any falling edge timing parameters.

I am attaching plots obtained for no transition(correct pin order) and no transition(incorrect pin order)

Thank you for your time and help.

<img width="1200" height="700" alt="Image" src="https://github.com/user-attachments/assets/d34720b6-a6a1-4f60-a42a-4e4682b84a1b" />

<img width="1200" height="700" alt="Image" src="https://github.com/user-attachments/assets/bc0a3f13-aec2-4ebb-a97e-8463e3751f22" />

sergeiandreyev, 2025-08-07:
Hi @adarshvp123, our standard cell developer is currently on leave, so we’ll need to wait for their response.

b10346, 2026-08-29:
This flip-flop is in obsolete category and should not be used. By the way the cell was checked from our site and it works. The problem could be due to port-order update in 2025 and old simulation setup using. Does the issue still exist?
(the issue need to be re-checked and closed)