IHP-GmbH/IHP-Open-PDK #648: Usage of the substrate 40/0 layer and digisub 60/0 layer

d-m-bailey, 2025-08-20:
@KrzysztofHerman @sergeiandreyev

Could you explain the usage cases and differences between the `substrate 40/0` layer and the `digisub 60/0` layer? 
They are both defined as "Substrate recognition layer for LVS" in the `SG13G2_os_layout_rules.pdf` file.

Currently, the klayout rules use `digisub 60/0` to create virtual substrate regions that do not connect to adjacent substrate regions. 
I'm not sure of the `substrate 40/0` intended usage, but currently it blocks the extraction of almost all devices (`cap_cmim` is allowed).

However, in the magic extraction rules, `substrate 40/0` is used to create the isolated substrate regions and `digisub 60/0` is not defined.

@FaragElsayed2 @RTimothyEdwards @H-Ojiro

KrzysztofHerman, 2025-08-20:
@d-m-bailey 
The 'substrate 40/0' is the default p-type substrate and the `digisub 60/0` layer is used for mixed signal circuits. 
AFAIK all originates with the introduction of the `ptap1` device, which is used in analog/RF designs, which can bring troubles in a case of a layout where you have analog and digital cells.  In our standard cells we do not provide tap cells since the `VSS` and `VDD` are connected to the `bulk` and `nwell` respectively.  LVS recognizes this connections as shorts and if you place on the same layout the `ptap1` device you can get conflict with the LVS. That is why you should use `digisub' layer to enclose the digital part of the layout.   

Here you can see two examples of the connection to the bulk in a standard cell using: a) ptap device and b) using silicide
a) in `sg13g2_inv1` cell the layers at the bottom are the same like in a `ptap1` device, active layer of nmos and under VSS strap are separated

[image]
  
b) in `sg13g2_tiehi` cell the active of the `nmos` devices and under the VSS strap are connected what implies a conductive silicide layer over it (duplicated with metal connection to both, what not always is a case). 

[image]

Long story short the introduction of these two layers permits to provide some more information to the  LVS  deck in a case of mixed signal designs.

d-m-bailey, 2025-08-20:
@KrzysztofHerman thanks for the explanation. I think I understand the issue of substrate shorts of different VSS voltages through p-diffusion over psubstrate and the use of these layers to virtually separate regions.

The question I had is why are two layers needed? They should both provide the same functionality, correct?

b10346, 2026-01-14:
" If either the substrate (40/0) layer or the sub! text is missing, a ptap1 device should not be extracted, correct? Also, the substrate (40/0) should not be used anywhere except ptap1 devices, right?"
Correct, right.

"My understanding is that while this allows the verification that all the ptaps are connected, it does not allow the checking of which ptaps are closest to a given nmos. This is because all of the nmos outside of digisub (60/0) (or nBuLay) have a bulk connection of sub!."
This is correct!

The Digisub layer is used in cases where the design contains both digital and analog blocks with ptap1 p-cells.
Digital blocks must be covered by the Digisub layer.

MOSFETs in digital gate schematics do not have ptap1 p-cells, so the bulks of the devices are connected directly to VSS.
This causes an LVS conflict, where two different nets (VSS and sub!) appear to be shorted.
The purpose of the Digisub layer is to indicate to LVS that this is not a real collision.

As far as I remember, the substrate layer is used in some p-cells to define a sub! terminal.