IHP-GmbH/IHP-Open-PDK #655: Isolated ESD diodes are missing

simonwaid, 2025-08-27:
The isolated ESD diodes (idiodevdd_2k, idiodevdd_4k, idiodevss_2k, idiodevss_4k) as documented here:

`https://ihp-open-pdk-docs.readthedocs.io/en/latest/verification/lvs/04_07_esd.html`
They are missing in the spice library and the ESD pcell. They would be useful for low-leakage, guarded input pins. 

It would be great to have the spice model. The lack of their presence in the pcells can be worked around, as there is a gds file of them in the test cases.

sergeiandreyev, 2025-08-29:
Hi @simonwaid, it should be cleaned-up, these devices are obsolete - were rarely used and we do not have proper models for them.

simonwaid, 2025-09-01:
@sergeiandreyev I hope that with cleaned-up, you don't mean removed. I wouldn't know how else to design an ESD protection where the input pin needs to be guarded. But maybe you have an idea. Anyway, I'm using them, so +1 on he rarely used.

sergeiandreyev, 2025-09-01:
@simonwaid, in fact it was exactly what I was planning to do 😅 - remove all left-over references to `idiode*` devices
these device layouts are not part of `sg13g2_pr` library anymore, and the LVS testcases should be cleaned up to not cause confusion
Maybe at some point we will reconsider to include them in OpenPDK, but we will need to have the model to begin with

> I wouldn't know how else to design an ESD protection where the input pin needs to be guarded

Maybe @b10346 can help here?

Scafir, 2025-09-01:
Hi, I just wanted to mention that I am also quite interested in the isolated ESD diodes for my application (femto ampere current sensing for CERN).
An official gds in sg13g2_io or sg13g2_pr would be especially useful, as it would enable different teams to be able to compare measurements of critical values for a common device. In my case, I do not really mind the missing or incomplete models: as long as I can experimentally measure the leakage to be low enough, it means my project is technically feasible. I can live with reduced or more uncertain ESD behavior.