IHP-GmbH/IHP-Open-PDK #656: New DRC deck - inconsistencies between marker database window and terminal output

KrzysztofHerman, 2025-08-27:
@FaragElsayed2 @atorkmabrains 

While I ran DRC on [this design](https://github.com/PhillipRambo/TO_July2025/blob/60_ghz_mpa/60_GHZ_MPA/design_data/gds/FMD_QNC_60_ghz_mpa_filled.gds) I got no DRC's reported in the marker database browser 

[image]

however in the log window and in the terminal I got violations

```
Number of DRC errors for maximum rule set: 0
2025-08-27 12:43:21 +0200: Memory Usage (625096K) : KLayout DRC run for maximum ruleSet completed in 100.23 seconds
27-Aug-2025 12:43:21 | ERROR   | ==================================================================================
27-Aug-2025 12:43:21 | ERROR   | ❌ --- KLayout DRC Check Failed: Violations were detected in the layout. --- ❌
27-Aug-2025 12:43:21 | ERROR   | ==================================================================================
27-Aug-2025 12:43:21 | ERROR   | Violated rules are : {'GFil.i', 'CntB.h1'}

27-Aug-2025 12:43:21 | INFO    | Total DRC Run time: 103.02 seconds (including execution, analysis, and reporting)
```

FaragElsayed2, 2025-08-27:
Hi @KrzysztofHerman 

Could you please explain how did you get this?

I ran it on my side and here is the results

GUI:

[image]

Batch:

[image]

[image]

sergeiandreyev, 2025-08-27:
Hi @FaragElsayed2, it seems to be user env related - I have the same results as you

KrzysztofHerman, 2025-08-27:
@FaragElsayed2 @atorkmabrains @sergeiandreyev 

I have checked the `Enable Antenna Checks` and it seems that the deck generates two `*lyrdb` files so the marker database viewer does not load any. When unselecting antennas all is as it should be

sergeiandreyev, 2025-11-10:
Hi @KrzysztofHerman, could you please check the fix on your side to confirm? So we can close the issue

atorkmabrains, 2026-01-15:
@sergeiandreyev and @KrzysztofHerman No change is required from our side.

PR #659 (2025-08-28): Klayout DRC Updates
- Fixes #656 
    - Cleaning up the antenna rule deck placeholder; this file will be updated once antenna checks are supported.

- Fixes #654
   - Supporting using spaces in the drc options for GUI run