IHP-GmbH/IHP-Open-PDK #667: LVS Bugs

p-fath, 2025-08-30:
Before starting the layout of my design, I tried to verify the LVS behaviour, however I've found some troubles here:

- [ ] 2x ptab placed in schematic (w=10u, l=0.3u) and in layout (pcell), connected together in layout with pin psub and pin vss. In layout they are connected together with M1.drawing and a pin vdd is placed on layer M1.text. LVS recognizes the pins of layout and schematic, but does not throw an error while substrate is connected to vdd, in schematic to vss?
[image]

- [ ] when the 2 symbols of ptab are both fliped an connected to the complement pin in the schematic but in layout everything is kept the same, there is no difference behaviour in after running LVS?

[image]

p-fath, 2025-08-30:
Same for the resistor rppd. LVS does not pass. 

[image]

[rppd_eval.zip](https://github.com/user-attachments/files/22058903/rppd_eval.zip)

d-m-bailey, 2025-09-01:
@p-fath The current klayout rule set does not check pin names. It just connects topology. I think the capability to check pin names (or at least a helper function to do so) was implemented in klayout 0.30.2.

We're working on a rule change that will flag schematic ports that aren't matched in the layout.

This is the answer to your first 3 comments.

Your 4th comment about hand drawn deep nwell ntap not being extracted as ntaps is because only ntaps that have the `well` label on layer `63/0` are recognized as devices. The connectivity is there whether the ntap device is recognized or not. For ptap on the hbt device, you'll need the `sub!` text to extract that as a `ptap` device.

Resistors have recently had a third substrate terminal added. The extraction rules may not handle this yet. Can't be sure unless we know the pdk version that you are using. The resistors schematic symbols may have been recently changed to ignore the spiceprefix parameter. I think klayout is expecting an `R` device in the schematic spice. Do you have an `X` device now?

sergeiandreyev, 2026-08-25:
Hi @p-fath, could you please confirm if the issues are fixed from your side?