IHP-GmbH/IHP-Open-PDK #674: Transient simulation converges on one computer but not on another

mohemza, 2025-09-03:
I have run the following netlist using ngspice on my personal laptop and the simulation works fine. When I run the same netlist on the external servers I run into timestep too small error. On both my laptop and the server I am using ngspice v44+.

[netlist_and_data.zip](https://github.com/user-attachments/files/22118998/netlist_and_data.zip)

dwarning, 2025-09-03:
Don't know your symbol. Write after NX=1 selft=0.
Switching off selfheating let the circuit simulate.

mohemza, 2025-09-04:
I added the selft=0 option and the note "The temperature limiting function received NaN" disappeared. However I still get a timestep too small error and my circuit does not simulate

dwarning, 2025-09-04:
This I got after few minutes:
[image]

sergeiandreyev, 2025-11-10:
Hi @mohemza, do you have any update on this issue?