IHP-GmbH/IHP-Open-PDK #675: Full DRC deck not yet implemented in klayout

DiarmuidSLICE, 2025-09-07:
Hello,

The following DRCs have not yet been implemented in klayout:

$PDK_ROOT/$PDK/[libs.tech/klayout/tech/drc/MissingRules_maximal.md](http://libs.tech/klayout/tech/drc/MissingRules_maximal.md)

There are ~ 100 of them (e.g most DNW related DRCs have not been implemented). 

Magic currently has a much more complete deck implemented. Same needs to be done for klayout to prevent users submitting "dirty" GDS files. 

Creating this ticket to raise / track progress on this issue.

Regards,

Diarmuid

sergeiandreyev, 2025-11-10:
Hi @DiarmuidSLICE, the full DRC deck is available now in KLayout

closed 2025-11-13 by sergeiandreyev

DiarmuidSLICE, 2025-12-02:
Thanks @d-m-bailey . No such thing as too much explanation:-)!

I did as you said and seems to have worked well. However, some new alerts appear after I enter git-status. 
Attached details this. Should I be concerned?

Thanks,

[updating_pdk.pdf](https://github.com/user-attachments/files/23879906/updating_pdk.pdf)

DiarmuidSLICE, 2025-12-03:
@d-m-bailey Perfect! Thanks for the help!

d-m-bailey, 2025-12-06:
@DiarmuidSLICE `git pull` includes `git fetch`. However, I like to check the changes after `git fetch` before I merge them, so I use the 2 step process.