IHP-GmbH/IHP-Open-PDK #685: Inductors in the PDK

NooriDan, 2025-09-13:
Hi, I've been trying to find the inductor models in the PDK but haven't been successful. Does the pdk have inductor models?

d-m-bailey, 2025-09-13:
@NooriDan my understanding is that the simulation inductor model is way too complicated to be simulated by a generic model using the the 4 parameters of the xschem symbol (nr, d, s, w). I've heard mention of the FastHenry tool in discussions to extract an equivalent spice model for a given layout, but haven't actually done anything myself. Please let us know if you discover a solution.

NooriDan, 2025-09-14:
I see, this makes sense. I did notice the pcells are available in Klayout but not in xschem. I will dig into this more and let you know.

joamatab, 2025-09-14:
Hi,

I made a PR that includes GDSFactory support for this PDK

once that gets merged you can use 

```
import gdsfactory as gf
c = gf.components.spiral_inductor()
c.show()
```

https://gdsfactory.github.io/gdsfactory/components.html#gdsfactory.components.spirals.spiral_inductor

sergeiandreyev, 2026-09-21:
> I will dig into this more and let you know.

Hi @NooriDan, do you have any update?

NooriDan, 2026-09-21:
HI @sergeiandreyev, I've used AI agents to help me fix this issue. 

They essentially can make the spiral inductor layout, run OpenEMS simulations and extract the Q factor, Inductance, and the full transfer function of the inductor for you. You can then model the inductor with RLC equivalent circuit if needed. Not sure if IHP pdk has added inductors for the PDK.

sergeiandreyev, 2026-09-22:
> Not sure if IHP pdk has added inductors for the PDK.

in the meantime we've added inductors symbols in xschem:

[image]

No lumped models released so far..