IHP-GmbH/IHP-Open-PDK #698: Cell_footprint attribute in Liberty models.

b10346, 2025-09-29:
Currently attributes "cell_footprint" are not consistent with cell names.
We propose to make update.
The proposal is following:

1) The name of “cell_footprint” attribute should be a root of the cell name without technology type and drive strength prefixes.

Example:
cell (“sg13g2_dfrbp_1”) {
....
cell_footprint:  “dfrbp” ;
....

2) Functional cells with equal functionality and different drive strength will have the same footprint (family).
3) Delay cells will have unique cell_footprint attribute.
4) Filler cells have cell_footprint = "fill"

We would like to discuss it community: please, any comments or feedback. If this approach is OK, we will implement this update.

Regards,
Alexey

sergeiandreyev, 2026-08-28:
@b10346, this should be updated, am I right?

closed 2026-08-28 by sergeiandreyev

b10346, 2026-08-29:
@sergeiandreyev you are right